RWTH-EBC/AixOCAT #36: Calibration function for temperature sensors

msc-d1s7fa1eza, 2019-03-27:
## What is the issue/what feature do you want to develop?

- Feature request: Add a calibration function for temperature sensors.

## What is the current application?

- A calibration function was used in a previous Master Thesis in combination with a Modelica model.

## What are the requirements?

- ...

## How do we want to solve it?

- Add the related function to this library.

## Priority, deadline?

- normal, 2019/05/31

katharinabr, 2020-11-03:
There are already functions in OSCAT. Check this before implementing new functions.

StGoebel, 2021-02-23:
@PhOstmann did you check the oscat functions?

PhOstmann, 2021-03-01:
The CALIBRATE function in OSCAT works pretty well if implemented correctly.

I already use CALIBRATE to calibrate several differential pressure sensors on test bench startup.

Other function blocks did not seem suitable.

closed 2024-01-02 by StGoebel