RWTH-EBC/AixOCAT #39: Bug in FT_PID

dfreiberger, 2020-07-01:
I believe there is a bug in FT_PID (this is also the same code as in OSCAT 3.33). If the TN value was previously not 0, and then is set to 0, it does not reset the the integral term and keeps accumulating if IEN was previously true. Is the intent that RST should be TRUE if TN = 0?

https://github.com/RWTH-EBC/AixOCAT/blob/8a28fdc201727e9d41ac7b74a69df654e50691a4/OSCAT/OSCAT/OSCAT/oscat_basic/POUs/Engineering/Control/FT_PID.TcPOU#L40-L44

dfreiberger, 2020-07-01:
Added commit to address issue by always resetting integrator if TN = 0, https://github.com/dfreiberger/AixOCAT/commit/d8fbb9a371aa5e538c4bfc888e2fbcf68efef849

alexanderAKU, 2020-07-02:
We changed the integrator reset to TRUE in the file `FT_PID.TcPOU` according to the proposed fix.

@dfreiberger thank you for your issue.

closed 2020-09-16 by alexanderAKU

dfreiberger, 2020-09-16:
@alexanderAKU thank you for resolving this!