RWTH-EBC/AixOCAT #65: Add the new Function Block TON_P

elconfa, 2021-03-04:
I would like to add a Function:
TON pausable. I hope this is useful for someone

FUNCTION_BLOCK TON_P	// Ton pausabile
VAR_INPUT
    IN : BOOL;
    PT : TIME;
    PAUSE : BOOL;
END_VAR
VAR_OUTPUT
    Q : BOOL;
    ET : TIME;
END_VAR
VAR
    rtPause : R_TRIG;
    tTimePaused : TIME;
    ton : TON;
END_VAR

IF NOT IN THEN
    tTimePaused := T#0S;
END_IF
rtPause(CLK := IN AND PAUSE);
IF rtPause.Q THEN
    tTimePaused := tTimePaused + ton.ET;
END_IF
ton(IN := IN AND NOT PAUSE, PT := PT - tTimePaused);
Q := ton.Q;
ET := tTimePaused + ton.ET;

[TON_P.zip](https://github.com/RWTH-EBC/AixOCAT/files/6081872/TON_P.zip)

msc-d1s7fa1eza, 2021-03-04:
Hello elconfa and welcome to the AixOCAT community!

Thank you very much for your suggestion. As I can see, your code pretty much follows, what was suggested in this thread (https://stackoverflow.com/questions/34204045/pausing-timers-in-structured-text), hence I do not see an issue integrating it.
However, I do recommend to follow the related workflow. Therefore, I kindly ask you to add the POU directly into the library (my suggestion is to add it within https://github.com/RWTH-EBC/AixOCAT/tree/develop/OSCAT/OSCAT/OSCAT/oscat_basic/POUs/Time%26Date ) and create a merge request. You should have permissions to create a feature branch from the develop branch. We will take care of your merge request then ;)

elconfa, 2021-03-04:
"As I can see, your code pretty much follows, what was suggested in this thread 
 https://stackoverflow.com/questions/34204045/pausing-timers-in-structured-text),"

Yes, it is. You are right. It is a lot of time that I use this code and I think that may be useful for other people.
I tried to upload the POU, but I have no experience using GIT. 

This is the response of the upload function:
Uploads are disabled.
File uploads require push access to this repository.

I'm sorry

PR #66 (2021-03-07): Add TON_P Function Block #65
Closes #65
closed 2021-03-07 by msc-d1s7fa1eza