RWTH-EBC/AixOCAT #70: Add FIWARE compatibility for MQTT-pyADS-Interface

rocketrocksyou, 2021-04-14:
## What is the issue/what feature do you want to develop?

- plug-and-play parser selection
- FIWARE parser for existing interface

## What is the current application?

- connection between FIWARE and TwinCAT

## How do we want to solve it?

- parser class for FIWARE

Maghnie, 2021-05-26:
Hi Florian, that's an interesting suggestion. How can it be different from issue #52 ?

FlRedder, 2021-05-26:
Hey @Maghnie, during our discussions with @tstorek we came to the conclusion that replaceable parsers (e.g. SimpleJSON, FIWARE) would be an enhancement to #52. We also need this feature for our implementation.

Maghnie, 2021-05-26:
Good to know! @msc-d1s7fa1eza and @StGoebel are finishing up #52 and merging it into development, so maybe then someone(?) can add the FIWARE features 🙂

StGoebel, 2021-05-26:
Hi @FlRedder, you can find the MQTT-pyADS-Interface in the development branch now. 
There are two parsers integrated you can use (SimpleJson and Influx).
Feel free to add further parsing functions and do not hesitate to ask if you have further questions.
Please start with a new branch if you start with further improvements.

StGoebel, 2021-05-26:
You can find further information here: https://github.com/RWTH-EBC/AixOCAT/wiki/mqtt-ads-interface